robotics-in-concert/rocon_composer #25: ROCON composer blockly 설치 중 오류

hughie, 2015-03-25:
ROCON composer blockly를 1.234.83.65서버에 설치하려고 하는데 npm install을 하니 아래와 같은 에러가 생깁니다.

readme.md에 있는 그 위의 과정은 이상없이 수행했습니다.

--------------------------------------
../src/libxmljs.cc: In function ‘v8::Handle<v8::Value> libxmljs::ThrowError(const char*)’:
../src/libxmljs.cc:98:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make: *** [Release/obj.target/xmljs/src/libxmljs.o] Error 1
make: Leaving directory `/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/roslib/node_modules/xmlshim/node_modules/libxmljs/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Linux 3.13.0-46-generic
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/roslib/node_modules/xmlshim/node_modules/libxmljs
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok

> contextify@0.1.13 install /home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/roslib/node_modules/xmlshim/node_modules/jsdom/node_modules/contextify
> node-gyp rebuild

child_process: customFds option is deprecated, use stdio instead.
make: Entering directory `/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/roslib/node_modules/xmlshim/node_modules/jsdom/node_modules/contextify/build'
  CXX(target) Release/obj.target/contextify/src/contextify.o
  SOLINK_MODULE(target) Release/obj.target/contextify.node
  SOLINK_MODULE(target) Release/obj.target/contextify.node: Finished
  COPY Release/contextify.node
make: Leaving directory `/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/roslib/node_modules/xmlshim/node_modules/jsdom/node_modules/contextify/build'
npm ERR! Linux 3.13.0-46-generic
npm ERR! argv "node" "/usr/local/bin/npm" "install"
npm ERR! node v0.12.0
npm ERR! npm  v2.7.1
npm ERR! code ELIFECYCLE

npm ERR! canvas@1.1.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the canvas@1.1.3 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the canvas package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls canvas
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/rocondev/rocon/rocon_composer/rocon_composer_blockly/npm-debug.log

eskim, 2015-03-25:
roslibjs 의존을 package.json 에서 제거하였습니다.
git pull 하신 후에 npm install 하시면 될 것입니다.

hughie, 2015-03-25:
네 말씀하신대로 하니 잘 되네요. 감사합니다.

closed 2015-03-25 by hughie