robotics-in-concert/rocon_composer #30: 환경 변수를 config.json으로 변경

dwlee, 2015-04-16:
기존에 composer 동작에 필요한 파라미터들을 환경변수로 등록하여 사용하던것을 
config.json 저장하고, 이를 node에서 읽어와 사용할 수 있도록 변경

다음과 같이 기존에 사용하던 환경변수를 변경
```
ROCON_COMPOSER_BLOCKLY_SERVER_PORT : The port to run web interface.
```
-> 엔진 분리로 인해 더 이상 사용하지 않음

```
ROCON_COMPOSER_BLOCKLY_MONGO_URL : mongodb connection URL 
(eg. mongodb://localhost:27017/rocon_composer)
```
-> "mongo_url": "< mongodb url >"
(default: mongodb://localhost:27017/rocon_composer)

```
ROCON_COMPOSER_BLOCKLY_ENGINE_SOCKET_URL : workflow engine's websocket url
```
-> 엔진 분리로 인해 더 이상 사용하지 않음

```
MSG_DATABASE : message database server address 
(eg. http://localhost:10000)
```
-> "rocon_protocols_webserver_address" : "< rocon protocol webserver address >"
(default: http://localhost:10000)

```
ROCON_COMPOSER_BLOCKLY_LOG_LEVEL : log level 
(eg. info, debug)
```
-> "log_level" : "< log level >"
(default: info)

```
ROCON_COMPOSER_BLOCKLY_SERVICE_REPO_BASE :  base github service repository 
(eg. eskim/test_ros_packages)
```
-> "service_repo_base" : "< git repo >"
```
ROCON_COMPOSER_BLOCKLY_SERVICE_REPO : forked github service repository 
(eg. waypoint/test_ros_packages)
```
-> "service_repo" : "< git repo >"

```
ROCON_COMPOSER_BLOCKLY_GITHUB_TOKEN : github access token to commit forked repository (Settings > Applications > Generate New Token)
```
->  "github_token" : "< github token >"

eskim, 2015-04-17:
작업 완료. 확인 하시고 close 시켜주세요

dwlee, 2015-04-17:
확인하였습니다. 
감사합니다.

closed 2015-04-17 by dwlee