robotics-in-concert/rocon_composer #33: 서비스 업로드시 master이 외의 branch로 업로드 할 수 있도록 개선

dwlee, 2015-04-16:
서비스 저장시 사용되는 repo 에서 무조건 master branch 만 사용하는것이 아니라 선택적으로 다른 branch를 사용할 수 있도록하고, 이를 설정하는 환경 변수 추가

eskim, 2015-04-23:
작업완료.

config.json 에 `service_repo_branch` 값에 branch 정보를 설정해 주셔야 합니다.

dwlee, 2015-04-23:
다음과 같은 문제가 있습니다. 
상황은 다음과 같습니다. 
git에 기본 패키지가 master로 설정되어 있고, config 설정에 common_service로 되어 있는 경우, 
local temp repo 에 git pull 을 하면 처음 브랜치는 master로 설정되어 있습니다. 
그래서 merge 할때 다음과 같이 에러가 발생합니다. 
```
tmp repo root /tmp/service_repository
merge  common_service
error: failed to get packages : [Error: Reference 'refs/remotes/common_service/HEAD' not found] Error: Reference 'refs/remotes/common_service/HEAD' not found
    at Error (native)
```

하지만 git checkout 을 통해 local temp repo 의 브랜치를 common_service 로 변경하면 이러한 문제가 없습니다. 
```
tmp repo root /tmp/service_repository
merge  common_service
```

이렇게 local temp repo의 기본 브랜치를 config 에 따라 바꾸어야 할것 같습니다.

eskim, 2015-04-23:
`/tmp/service_repository` 를 완전히 지우고 다시한번 테스트 해보시고 같은 문제가 발생하는지 알려주세요~

dwlee, 2015-04-23:
확인하였습니다. 
이슈는 닫도록 하겠습니다.

closed 2015-04-23 by dwlee