robotics-in-concert/rocon_composer #38: merge  undefined

dwlee, 2015-04-23:
다음과 같은 에러가 발생합니다. 
확인 부탁드립니다. 

```
tmp repo root /tmp/service_repository
https://github.com/dwlee/rocon_service_packages.git
merge  undefined
error: failed to get packages : [Error: String shorthand is required.] Error: String shorthand is required.
    at Error (native)
    at /home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/node_modules/nodegit-promise/lib/node-extensions.js:25:10
    at doResolve (/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/node_modules/nodegit-promise/lib/core.js:122:5)
    at new Promise (/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/node_modules/nodegit-promise/lib/core.js:102:3)
    at /home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/node_modules/nodegit-promise/lib/node-extensions.js:17:12
    at Function.<anonymous> (/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/lib/utils/lookup_wrapper.js:28:12)
    at Repository.getReference (/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/lib/repository.js:127:20)
    at Repository.getBranch (/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/lib/repository.js:91:15)
    at Repository.mergeBranches (/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/lib/repository.js:668:10)
    at /home/rocondev/rocon/rocon_composer/rocon_composer_blockly/src/service_store.js:65:23
    at /home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/node_modules/nodegit-promise/lib/core.js:64:15
    at flush (/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/node_modules/nodegit-promise/node_modules/asap/asap.js:27:13)
    at process._tickCallback (node.js:355:11)

```

eskim, 2015-04-23:
문서가 업뎃이 안되었네요. 

config.json 파일에 

    "service_repo_branch": "master"

라인을 추가해 주시면 에러발생하지 않을 것입니다.

dwlee, 2015-04-23:
확인하였습니다. 
에러가 발생하지 않았습니다. 
이슈는 닫겠습니다.

closed 2015-04-23 by dwlee