robotics-in-concert/rocon_composer #44: 저작된 서비스 export시 pull request 되지 않는 문제

dwlee, 2015-05-06:
저작된 서비스 export 하면 미리 지정된 git 경로에 pull request 됩니다. 
하지만 이 기능이 동작하고 있지 않습니다. 

rocon composer 쪽 로그
```
---------------- .launcher --------------------
<launch>
  <node pkg="concert_common_services" name="publish_world" type="publish_world.py">
  </node>
  <node pkg="concert_workflow_handler_blockly" name="workflow_handler_blockly" type="workflow_handler.py">
  </node>
</launch>
---------------- .service --------------------
name: pickup_delivery_service
description: pickup delivery service test
author: dwlee
priority: 10000
interactions: concert_common_test_services/pickup_delivery_service
parameters: concert_common_test_services/pickup_delivery_service
launcher_type: roslaunch
launcher: concert_common_test_services/pickup_delivery_service

{ '0': undefined }

```

저작된 서비스를 pull request 하는 git 경로 
```
https://github.com/dwlee/rocon_service_packages
```

config.json 정보 

```
{ port: 9999,
  mongo_url: 'mongodb://localhost:27017/rocon_authoring',
  rocon_protocols_webserver_address: 'http://localhost:10000',
  log_level: 'info',
  service_repo_base: 'dwlee/rocon_service_packages',
  service_repo: 'dwlee/rocon_service_packages',
  service_repo_branch: 'common_service',
  github_token: '122...' }

```

eskim, 2015-05-06:
github_token 은 122가 아닌 실제값을 입력하신게 맞죠? 확인해 보겠습니다

dwlee, 2015-05-06:
네 줄임으로 작성하였습니다.

eskim, 2015-05-06:
테스트시 주석처리 되었던게 커밋이 되었나봅니다.
수정하였습니다.

dwlee, 2015-05-06:
PR 이 정상적으로 됨을 확인하였습니다. 
이슈는 닫겠습니다.

closed 2015-05-06 by dwlee