robotics-in-concert/rocon_composer #54: service 저작후 export 시 에러 발생

dwlee, 2015-05-18:
서비스 export 시 다음과 같은 에러가 발생합니다. 

```
/home/rocondev/rocon/rocon_composer/rocon_composer_blockly/node_modules/nodegit/node_modules/nodegit-promise/node_modules/asap/asap.js:45
                throw e;
                      ^
Error: Reference 'refs/remotes/rapp_package/HEAD' not found
    at Error (native)
```

eskim, 2015-05-18:
제쪽에서는 재현이 안되네요. 혹시 다른 단서가 없나요?

dwlee, 2015-05-18:
로그를 보면```rapp_package``` 부분은 참고하지 못해서 에러가 난다고 합니다. 
하지만 rapp_package는 서비스 저작 쪽 branch가 아니라 rapp 저작에 관련된 branch 입니다. 
아마 브랜치 설정을 하는 부분이 꼬인게 아닐까요?

dwlee, 2015-05-18:
오탈자가 있는것 같습니다. 
https://github.com/robotics-in-concert/rocon_composer/blob/master/rocon_composer_blockly/src/generators/service.js#L226

eskim, 2015-05-18:
제 쪽에서는 rapp_repo_branch 와 service_repo_branch 설정이 같아 에러가 나지 않았었네요.
버그 찾아주셔서 감사합니다.!

수정했습니다.

closed 2015-06-03 by dwlee